bigtreetech/MMU2-DIP #4: Can not connect skr 1.4 turbo

pstit006, 2021-01-02:
I have tried and tried I cannot get the mmu2 dip to communicate with the skr 1.4 turbo

pstit006, 2021-01-05:
Can anybody help

XylenC4, 2021-01-06:
I've got it working.
[Configuration_adv.txt](https://github.com/bigtreetech/MMU2-DIP/files/5777667/Configuration_adv.txt)
[Configuration.txt](https://github.com/bigtreetech/MMU2-DIP/files/5777668/Configuration.txt)

I'm using the TFT Socket so SERIAL0. To use this port it has to be disabled for general G-Command usage
//#define SERIAL_PORT_2 0

last but not least you have to define the Port (Serial is SERIAL0)
 // #define MMU2_SERIAL_PORT 0
  #define MMU2_SERIAL Serial

Make sure you're serial connection is connected right!
[image]

pstit006, 2021-01-06:
Ok I will give that a try I'm running a tft70 and I watched a video for using the i2c serial as serial 3 so I could keep my tft

XylenC4, 2021-01-06:
If you only use MARLIN Mode on the TFT (as I do) it will work without Serial0

pstit006, 2021-01-16:
Ok I've been trying to get this board to work I'm using the btt mmu2-dip board with tmc2209 drivers I've tried connecting to the tft port and still not talking to the mmu2-dip

AJAX2500, 2021-02-06:
Hey guys. I spent a lot of time smashing my head against a wall on trying to get the MMU board to connect to any board running 2.0.7.2 and ultimately I found that in the version.h of the MMU firmware file there is the following line.

static const uint16_t fw_buildnr = 130; //!< number of commits preceeding current HEAD

Once I changed this to the following everything connected up and worked perfect.

static const uint16_t fw_buildnr = 132; //!< number of commits preceeding current HEAD

Hope this is helpful to you!

Ven827, 2023-09-12:
Hey, just having a similar issue, my serial port seems to be 3.3V while the MMU2 needs 5V, I plugged it into a different 5V source and it works but I wasn't able to get any communication to happen. I'm also not able to compile if I comment out #define MMU2_SERIAL_PORT 0. Any help would be appreciated.

XylenC4, 2023-09-12:
I'll haven't compiled Marlin for a while as I'll moved over to a SKR PICO and KLIPPER and a Palette 2+ so take my answer with a grain of salt...

You shouldn't comment it out but replace the #define MMU2_SERIAL_PORT 0 with #define MMU2_SERIAL Serial. The MMU will require a serial port for communication, so there is no sense on disabling the port by removing the line.

Use a USB -> Serial Port adapter to verify the TTL (5V) serial communication to the MMU and the communication to your main board (which one do you use).