bigtreetech/MMU2-DIP #6: MMU2 doesn't connect with GTR 1.0

av71966, 2021-01-10:
Hi. I tried to connect MMU2 to GTR 1.0 Board.  I connect it to UART3 (TFT connection)
MMU can reset from LCD menu, but don't load\uload filament and don't homing.
Marlin 2.0.7.4

Configuration 
#define SERIAL_PORT -1
#define SERIAL_PORT_2 3

Configuration_adv
  #define MMU2_SERIAL_PORT 3
  #define MMU2_SERIAL Serial3

What's wrong?

XylenC4, 2021-01-14:
check out this thread (even if you use another board / chipset)

https://github.com/bigtreetech/MMU2-DIP/issues/4

-> Disable #define SERIAL_PORT_2 3 and #define MMU2_SERIAL_PORT 3

av71966, 2021-01-14:
> check out this thread (even if you use another board / chipset)
> 
> #4
> 
> -> Disable #define SERIAL_PORT_2 3 and #define MMU2_SERIAL_PORT 3

Hi, i use UART 3 for TFT now. I tried to use serial Port 2 (ruspberry port) and have the same result...
If i want to use serial port 2 i need only define MMU2_SERIAL_PORT 2 and comment MMU2_SERIAL in Configuration ADV?

XylenC4, 2021-01-14:
You will need a separate serial connection to get this working. 

//#define SERIAL_PORT_2 3
//#define MMU2_SERIAL_PORT 3
#define MMU2_SERIAL Serial2

I'm using the TFT but only in MARLIN MODE so no Serial Connection is needed for my TFT!

I recommend plugging in a Serial to USB connection to check if the connection is working. On Reset you will see an X0 sended by MARLIN.

av71966, 2021-01-21:
I fix problem.
Update marlin to new bugfix. Now it work perfect on uart6
https://github.com/MarlinFirmware/Marlin/pull/20771

closed 2021-01-21 by av71966

pstit006, 2021-01-21:
How did you get the mmu2-dip to work can you provide a configuration and can you share a picture of how you have it hooked up I still can't seem to get this board working

Montezuma-69, 2021-05-18:
Av71966 Same problem here. 
#define MMU2_SERIAL Serial3 
works with TFT port

but

#define MMU2_SERIAL Serial6
does not work on USART6 at the WIFI port.

What did you set in configuration_adv.h?
Thank you so much