bigtreetech/MMU2-DIP #10: firmware.hex

nonaak, 2021-04-25:
where can i find firmware.hex?
or do i have to make it my self in platformIO?

Marlin display say "update mmu firmware" or do i somthing wrong? i folow the youtube of Chris Riley

thanks

ellensp, 2022-03-26:
Marlin has function 
```CPP
/* 
* Check if MMU has compatible firmware
 */
void MMU2::check_version() {
  if (buildnr < MMU_REQUIRED_FW_BUILDNR) {
    SERIAL_ERROR_MSG("Invalid MMU2 firmware. Version >= " STRINGIFY(MMU_REQUIRED_FW_BUILDNR) " required.");
    kill(GET_TEXT_F(MSG_KILL_MMU2_FIRMWARE));
  }
}

so it the firmware value  read form the mmu2 is less than MMU_REQUIRED_FW_BUILDNR
It will tell you to update the firmware

In Marlin MMU_REQUIRED_FW_BUILDNR is set with

`#define MMU_REQUIRED_FW_BUILDNR TERN(MMU2_MODE_12V, 132, 126)
`
Which says set MMU_REQUIRED_FW_BUILDNR to 132 if MMU2_MODE_12V is set
Otherwise set it to 126

This is set in MMU@-DIP source MMU2/src/version.h
`static const uint16_t fw_buildnr = 133; //!< number of commits preceeding current HEAD
`

ellensp, 2022-03-26:
You build the firmware from source. 
This is required to set the type of stepper driver and stepper driver current