bigtreetech/MMU2-DIP #13: Unable to upload firmware to board. Please help me.

newbie-0707, 2021-08-04:
MMU2-dip board is not port-aware.
Created "firmware.hex" file with Visual Studio Code program
You have currently installed the CH340 driver on Windows 10.
And when I buy the board, I connect it with my computer with the cable, so the device manager recognizes the port well.

Afterwards, the CH340 port is not detected even if you download and run the flyMCU program and press the EnumPort button

I tried to download STM32flash in another way and upload firmware with the command from the CMD window, but I get a "failed to init device." error message.

Install Ubuntu with Windows 10 Hyper-v to install the CH340 driver and try to install STM32flash without USB recognition.

Is the board broken? Or is it the wrong way I install it? I'd appreciate it if you could let me know.

It may sound strange because it's translated into a translator, but please.

newbie-0707, 2021-08-31:
Firmware upload was fixed
It does not work with stm32flash or Flymcu programs, so you can purchase ST-link, connect it to SWD, and upload it.

fjpg1970, 2023-04-20:
> Firmware upload was fixed It does not work with stm32flash or Flymcu programs, so you can purchase ST-link, connect it to SWD, and upload it.
> 
> If it is not possible to connect via ST-link, connect 3.3v power to BOOT0 pin and reset it (this is not certain information) BOOT0 is connected to GND through the C17 capacitor on the board.

I don't have st-link, will putting 3.3v in boot0 work?