bigtreetech/MMU2-DIP #14: unable to compile in PlatformIO

i8it, 2021-08-11:
I get several digital pin warnings and then I get a compile failure.  Is my board definition correct?

Here is the terminal window result:

> Executing task: C:\Users\Megamind\.platformio\penv\Scripts\platformio.exe run --environment MMU2_F030C8 <

Processing MMU2_F030C8 (platform: ststm32; board: MMU2_F030C8; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/MMU2_F030C8.html
PLATFORM: ST STM32 (14.1.0) > MMU2_F030C8
HARDWARE: STM32F030C8T6 48MHz, 8KB RAM, 64KB Flash
DEBUG: Current (jlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-arduinoststm32 4.10700.200103 (1.7.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Warning! Cannot find linker script for the current target!

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing TMCStepper @ >=0.6.2,<1.0.0
Library Manager: TMCStepper @ 0.7.1 has been installed!
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SPI> 1.0
|-- <IWatchdog> 1.0.0
Building in release mode
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Print.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WString.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\abi.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\board.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\hooks.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\itoa.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\main.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\new.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma2d.c.o
[... 218 lines cut ...]
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: uart.c:(.text.uart_init.part.0+0xf4): undefined reference to `PinMap_UART_RX'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/FrameworkArduino/stm32/uart.c.o: in function `uart_debug_init':
uart.c:(.text.uart_debug_init+0x48): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: uart.c:(.text.uart_debug_init+0x4c): undefined reference to `PinMap_UART_TX'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: uart.c:(.text.uart_debug_init+0x50): undefined reference to `PinMap_UART_RX'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/FrameworkArduino/stm32/uart.c.o: in function `uart_debug_write':
uart.c:(.text.uart_debug_write+0xd8): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: uart.c:(.text.uart_debug_write+0xdc): undefined reference to `PinMap_UART_TX'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/FrameworkArduino/wiring_digital.c.o: in function `pinMode':
wiring_digital.c:(.text.pinMode+0x80): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0x88): undefined reference to `PinMap_PWM'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/FrameworkArduino/wiring_digital.c.o: in function `digitalWrite':
wiring_digital.c:(.text.digitalWrite+0x38): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/FrameworkArduino/wiring_digital.c.o: in function `digitalRead':
wiring_digital.c:(.text.digitalRead+0x3c): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/src/src/HAL/SoftwareSerial.cpp.o: in function `SoftwareSerial::SoftwareSerial(unsigned short, unsigned short, bool)':
SoftwareSerial.cpp:(.text._ZN14SoftwareSerialC2Ettb+0xe8): undefined reference to `digitalPin'
c:/users/megamind/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/MMU2_F030C8/libd35/libSPI.a(SPI.cpp.o): in function `SPIClass::SPIClass()':
SPI.cpp:(.text._ZN8SPIClassC2Ev+0x44): undefined reference to `digitalPin'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\MMU2_F030C8\firmware.elf] Error 1
==================================================================== [FAILED] Took 29.78 seconds ====================================================================
The terminal process "C:\Users\Megamind\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MMU2_F030C8'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

pewe0001, 2022-02-22:
have you solved the problem?
i have the same and looking for al solving

ellensp, 2022-03-26:
I forced the platform version  to 12.1  vs the 14.1 your trying to use

Ie In platformio.ini

Find
```
[env:MMU2_F030C8]
platform  = ststm32
```

to
```
[env:MMU2_F030C8]
platform  = ststm32@~12.1
```

This compiles for me