bigtreetech/MMU2-DIP #16: Impossible Compilation

FNeo31, 2021-09-18:
I get this error and I can't compile.

Tried another versions of PlatformIO and framework-arduinoststm32 and anything result. 
Can anyone help?

```
Processing MMU2_F030C8 (platform: ststm32; board: MMU2_F030C8; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/MMU2_F030C8.html
PLATFORM: ST STM32 (14.0.1) > MMU2_F030C8
HARDWARE: STM32F030C8T6 48MHz, 8KB RAM, 64KB Flash
DEBUG: Current (jlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-arduinoststm32 4.10700.200103 (1.7.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Error: Missing PlatformIO build script C:\Users\fjpin\.platformio\packages\framework-arduinoststm32@4.10700.200103\tools\platformio\platformio-build.py!
========================================================================================================= [FAILED] Took 0.80 seconds =========================================================================================================The terminal process "C:\Users\fjpin\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

newbie-0707, 2021-10-30:
me too...i want compile...

newbie-0707, 2021-11-05:
Once the compilation was successful.

I'll just share with you how I did it. I hope you give it a try

Install another version of framework-arduinoststm32

Change the number after "platform_packages = framework-arduinoststm32@>=4.10700,<4.10800" in the platformio.ini file.

For example 4.10800, < 4.10900

After that, if you look in the tools folder in the newly installed framework-arduinoststm32 folder, there is a .flake8 file After copying this

Paste it in C:\Users\calca\.platformio\packages\framework-arduinoststm32@4.10700.200103\tools

Create a new folder in the tools folder, change the folder name to platformio, and put the platformio-build.py file in the platformio folder.

Add library in VScode PlatformIO IDE PIO home.
softwareserial
TMC stepper

Add Platforms
ststm32
atmel AVR (not sure)

newbie-0707, 2021-11-05:
Look up the contents of the platformio.ini set file

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = MMU2
boards_dir = buildroot/share/PlatformIO/boards
default_envs = MMU2_F030C8

[common]
default_src_filter = +<src/*>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
lib_deps = TMCStepper@>=0.6.2,<1.0.0
build_flags = -fmax-errors=5

[env:MMU2_F030C8]
platform = ststm32
extends = common
board = MMU2_F030C8
framework = arduino
platform_packages = framework-arduinoststm32@>=4.10700,<4.10800
build_flags = ${common.build_flags}
	-IMMU2/src/HAL -std=gnu++14
src_filter = ${common.default_src_filter}
extra_scripts = ${common.extra_scripts}
	buildroot/share/PlatformIO/scripts/generic_create_variant.py
	buildroot/share/PlatformIO/scripts/MMU2_F030C8.py
lib_deps = 
	${common.lib_deps}
	teemuatlut/TMCStepper@^0.7.3
	fysetc/SoftwareSerialM@^1.0.0
lib_ignore = SoftwareSerialM
upload_protocol = jlink
debug_tool = jlink

newbie-0707, 2021-11-05:
c_cpp_properties.json Set contents are also uploaded. I wish you refer to that

Please focus on includepath

//
// !!! WARNING !!! AUTO-GENERATED FILE!
// PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
// https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
//
{
    "configurations": [
        {
            "name": "PlatformIO",
            "includePath": [
                //"C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/include",
                "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/MMU2",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/IWatchdog/src",
                "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/.pio/libdeps/MMU2_F030C8/U8glib-HAL/src",
                "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/.pio/libdeps/MMU2_F030C8/TMCStepper/src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/SPI/src",
                "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/.pio/libdeps/MMU2_F030C8/SoftwareSerialM",
                "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/MMU2/src/HAL",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/avr",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32/LL",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32/usb",
                //"C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32/OpenAMP",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32/usb/hid",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino/stm32/usb/cdc",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Drivers/STM32F0xx_HAL_Driver/Inc",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Drivers/STM32F0xx_HAL_Driver/Src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/STM32F0xx",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src",
               // "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/OpenAMP",
                //"C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/OpenAMP/open-amp/lib/include",
                //"C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/OpenAMP/libmetal/lib/include",
                //"C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Middlewares/OpenAMP/virtual_driver",
                "C:/Users/calca/.platformio/packages/framework-cmsis/CMSIS/Core/Include",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/system/Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/gcc",
                "C:/Users/calca/.platformio/packages/framework-cmsis/CMSIS/DSP/Include",
                "C:/Users/calca/.platformio/packages/framework-cmsis/CMSIS/DSP/PrivateInclude",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/cores/arduino",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/variants/MMU2_F030C8",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/Wire/src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/EEPROM/src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/Keyboard/src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/Mouse/src",
                "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/Servo/src",
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "path": [
                    //"C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/include",
                    "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/MMU2",
                    "C:/Users/calca/.platformio/packages/framework-arduinoststm32@4.10700.200103/libraries/IWatchdog/src",
                    "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/.pio/libdeps/MMU2_F030C8/U8glib-HAL/src",
                    "C:/Users/calca/Downloads/MMU2-DIP-master/MMU2-DIP-master/firmware/.pio/libdeps/MMU2_F030C8/TMCStepper/src",
[... cut ...]
                ""
            ],
            "cStandard": "c11",
            "cppStandard": "c++14",
            "compilerPath": "C:/Users/calca/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/arm-none-eabi-gcc.exe",
            "compilerArgs": [
                "-mcpu=cortex-m0",
                "-mthumb",
                ""
            ]
        }
    ],
    "version": 4
}

After setting it up like this, I compiled it, but about 3 warnings appeared and the compilation was completed.

newbie-0707, 2021-11-05:
Also check the compile log.

I haven't written on the board yet. I'll leave it as a ripple for your reference.

> Executing task in folder firmware: C:\Users\calca\.platformio\penv\Scripts\platformio.exe run <

Processing MMU2_F030C8 (platform: ststm32; board: MMU2_F030C8; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/MMU2_F030C8.html
PLATFORM: ST STM32 (15.0.0) > MMU2_F030C8
HARDWARE: STM32F030C8T6 48MHz, 8KB RAM, 64KB Flash
DEBUG: Current (jlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-arduinoststm32 4.10700.200103 (1.7.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.3
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <IWatchdog> 1.0.0
Building in release mode
Compiling .pio\build\MMU2_F030C8\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Print.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\WString.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\abi.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\board.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\hooks.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\itoa.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\main.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\new.cpp.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma2d.c.o
Compiling .pio\build\MMU2_F030C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma_ex.c.o
[... 197 lines cut ...]
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC2130Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC2160Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC2208Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC2209Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC2660Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC5130Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMC5160Stepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\TMCStepper.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\bcm2835_spi.cpp.o
Compiling .pio\build\MMU2_F030C8\libab2\TMCStepper\source\bcm2835_stream.cpp.o
Compiling .pio\build\MMU2_F030C8\lib1cf\IWatchdog\IWatchdog.cpp.o
Archiving .pio\build\MMU2_F030C8\libab2\libTMCStepper.a
[... cut ...]
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  38.1% (used 3120 bytes from 8192 bytes)
Flash: [=======   ]  70.9% (used 46452 bytes from 65536 bytes)
Building .pio\build\MMU2_F030C8\firmware.bin
================================================================================================== [SUCCESS] Took 20.39 seconds ==================================================================================================

터미널이 작업에서 다시 사용됩니다. 닫으려면 아무 키나 누르세요.

FNeo31, 2021-11-05:
I had a problem on directory of stm32 library.

After that I have my board working.
Need to make some changes in the code to put the tool change more viable. And put the tool changes errors close to zero.

closed 2021-11-05 by FNeo31