bigtreetech/MMU2-DIP #18: MMU2-DIP will not reset on reset line

ellensp, 2022-03-27:
I have a brand new MMU2-DIP

I  connected to a FTDI interface to it since I was having issue getting it to communicate to do some diagnosis

FTDI -> P1 on MMU2-DIP
5v -> 5v
RX-TX
TX-RX
GND-GND

It powers on and stay "start" @ 115200 baud as it should.

When I press the reset button it resets as expected and the D6 LED lights up RED momentarily and It says start again over serial.

But when I connect the reset pin on P1 to GND momentarily, It does not restart, D6 LED does not light up and I don't get a start message,

The P1 reset line is require for the MMU2 to work.

Please advise.

I suspect I have a faulty board.

ellensp, 2022-03-30:
Please Confirm that a hardware reset of the MMU2-DIP by momentarily holding reset pin on P1 low is meant to work.
Switching to software reset and sending X0 over serial to MMU2-DIP resets as expected.

geheimo66, 2022-04-08:
for the reset, you have to remove the capacitor c15 at the reset button. then the reset works via p1.

closed 2022-04-08 by ellensp